EnviroDIY/YosemitechModbus #2: Test CHL sensor

aufdenkampe, 2017-07-18:


aufdenkampe, 2017-07-27:
My test output of the Chlorophyl sensor (Y514 Chlorophyl with brush) is discussed in issue #3.

SRGDamia1, 2018-02-01:
Can you verify timing for me for the Chloropyhll sensor?

I would like to know how fast it will respond to any command, how long after the sensor begins taking readings until something stable comes out, and how fast new readings can be returned.

aufdenkampe, 2018-02-01:
It worked!

Output from *ChangeSlaveID_AltSoftSerial.ino*, changing from 0x05 to 0x03, with debugging on.
```
ChangeSlaveID_AltSoftSerial.ino
Allowing sensor and adapter to warm up
10...9...8...7...6...5...4...3...2...1...

Changing Slave ID:
Raw Request >>> {0x05, 0x10, 0x30, 0x00, 0x00, 0x01, 0x02, 0x03, 0x00, 0xA4, 0x63}
Raw Response (8 bytes) <<< {0x05, 0x10, 0x30, 0x00, 0x00, 0x01, 0x0F, 0x4D}
Address change complete.
```

Output from *GetSlaveID_AltSoftSerial.ino*.
```
GetSlaveID_AltSoftSerial.ino
Allowing sensor and adapter to warm up
10...9...8...7...6...5...4...3...2...1...

Scan takes several minutes, please be patient!

Scanning for Yosemitech modbus sensors....
------------------------------------------
Modbus Address ------ Sensor Serial Number
     0x03      ------    YL3617022701
```

This was my bad, because I first ran ChangeSlaveID_AltSoftSerial.ino with the wrong model number, and although it said it failed, it actually succeeded in reseting the SlaveID, so when I ran it again with the correct model number, it didn't work because the SlaveID had already been changed.

aufdenkampe, 2018-02-01:
Here is where I shared some timing results for the Y514 sensor: https://github.com/EnviroDIY/YosemitechModbus/issues/3#issuecomment-316819583

I'll need to retest.

aufdenkampe, 2023-12-21:
This should have been closed a long time ago. The Y514 works well with this repo!

closed 2023-12-21 by aufdenkampe