EnviroDIY/YosemitechModbus #3: Test changing brush interval

aufdenkampe, 2017-07-18:
Old sensors versions, including those that Sara has, did not support "Activate Brush", "Get Brush Interval", "Set Brush Interval" commands. 

In the old sensor versions, the brush spins 3x one way and 1x the other way, as soon as the sensor gets power. It takes 10-11 sec to do. It will repeat every 30 minutes.

In the new sensor versions, the brush does not spin when the sensor is powered.

Look at code and notes in Get Values example sketch.

aufdenkampe, 2017-07-20:
I tested the new GetValues.ino on all three of my sensors, purchased in April 2017:
* Y511 Turbidity with brush
* Y514 Chlorophyl with brush
* Y520 Conductivity

Good news is that all give believable values and the brushes (for Y511 & Y514) activate with the "Activate Brush" command. The bad news is that for both Y511 & Y514, none of the other functions work.

I played around with the "Waiting for sensor and adapter to be ready" delay (line 77 in sketch). Increasing from the original 0.5 sec to 1.3 sec was necessary to get Y520 to work, as expected, but increasing up to 50 sec had no affect on getting responses for Y511 & Y514.  See example output below.

Note 1: The sketch returns a "Failed to activate brush!" error, but the brush actually does move exactly when the "Activating brush" command writes to the serial monitor.

Note 2: For Y514 their appears to be a ~10 sec delay from the first result value before a real value is transmitted. Because that chlorophyl sensor emits a blue light each time a measurement is made, it's clear to see that no lights are emitting in this time. I'm wondering if there is some kind of automatic delay after a brush is activated.

@SRGDamia1, any ideas?

```
Yosemitech Y511 sensor for Turbidity
Waiting for sensor and adapter to be ready.
Getting sensor version.
    Current Hardware Version: 0.00
    Current Software Version: 0.00
Getting sensor serial number.
    Serial Number: 
Getting sensor calibration equation.
    Current Calibration Equation: final = 0.00*raw + 0.00
Getting sensor cleaning interval.
    Sensor auto-cleaning interval: 0 minutes
Resetting cleaning interval to 30 minutes.
    Set interval failed!
Starting sensor measurements
    Failed to start measuring!
Allowing sensor to stabilize..
10...9...8...7...6...5...4...3...2...1...

Activating brush.
    Failed to activate brush!
Continuing to stabilize..
12...11...10...9...8...7...6...5...4...3...2...1...

Temp(°C)  Turbidity(NTU)
0.00      0.00
24.62      -4.65
24.62      -4.65
24.62      -4.65
24.62      -4.65
24.62      -4.65
24.62      -4.65
24.62      -4.65
24.62      -4.80
24.62      -4.24
24.75      1.62
24.75      4.49
24.75      11.85
24.75      14.01
24.75      18.97
24.75      31.36
24.75      37.09
```

SRGDamia1, 2017-07-21:
I'm sorry.  It worked fine on my Y511 except for the brushing command.  If the chlorophyll sensor failed to start measuring, then I'm not surprised you're not seeing the light go on.

Are you sure you have your sensor slave ID/address right?  Can you see the sensor reply with the "[GetSlaveID](https://github.com/EnviroDIY/YosemitechModbus/blob/master/utilities/GetSlaveID/GetSlaveID.ino)" utility sketch?

Try turning on the debugging and reprinting the output with all of the commands and responses printed out.  (Uncomment line 61 "sensor.setDebugStream(&Serial);")  Then you can try hooking the sensor to your computer and sending the commands through Simply Modbus Master.  That will also show you if the sensor is responding by the response isn't being parsed correctly or if there is no response at all.

You could also try to increase the values for "modbusTimeout" and "modbusFrameTimeout" at the very bottom of YosemitechModbus.h to see if that helps.  The modbusFrameTimeout length is defined by the modbus protocol and _should_ be correct, but the time it takes the sensor to respond (modbusTimeout (500ms)) might be inconsistent between the sensors.  I think in my tests it was 300-400ms for my sensors, but maybe 500 isn't a big enough buffer.

My last suggestion would be change the "stream->read();" in the "emptyResponseBuffer" function with something like "char c = stream->read(); _debugStream->print(c);" to see if somehow data is getting dumped out of the buffer at the end of the timeouts.

aufdenkampe, 2017-07-27:
@SRGDamia1, thanks for your ideas. 

I did use the "GetSlaveID" utility sketch (forgot to mention it), which returned the SlaveID but not the sensor serial number for the Y511 & Y514 sensors (but it returned both for the Y520 sensor).

I'll work on trying some of your other ideas this afternoon.

aufdenkampe, 2018-01-23:
**Update**, after substantial changes to this library, especially it's dependence since Sep. 6, 2017 on the new EnviroDIY [SensorModbusMaster](https://github.com/EnviroDIY/SensorModbusMaster) library.

When trying the current and much revised [GetValues](https://github.com/EnviroDIY/YosemitechModbus/tree/master/examples/GetValues) example sketch and the [GetSlaveID](https://github.com/EnviroDIY/YosemitechModbus/tree/master/utilities/GetSlaveID) utility sketch on the Y520 Conductivity, Y511 Turbidity & Y514 Chlorophyl sensors, I have continued to have challenges:
*  Using the [hardware setup from July](https://github.com/EnviroDIY/YosemitechModbus/blob/master/doc/HardwareConfigPhoto.jpg), the Y520 worked for both sketches, but Y511 & Y514 did not.
* After following suggestions by @SRGDamia1 to either increase the voltage to >5V and/or try AltSoftSerial rather than SoftwareSerial, I got the following results:
  * Both old and new conductivity sensors responds on 5V, using either SoftwareSerial or AltSoftSerial.
  * The new DO sensor (from @fisherba) responds on 5V, using AltSoftSerial but not SoftwareSerial.
  * My old & new turbidity and chlorophyll sensors, both with brushes, don't respond on 5V, even on hardware Serial1, but do on 9V.
* It seems that although I can now *read* registers containing SlaveID, Serial Number and data, I am still not able to *write* registers to ChangeSlaveID or to SetBrushInterval. In fact, for brush interval, it seems like the effort set the interval results in rewriting it to 0 minutes, as the brush auto-activates the first time it is powered, but then never again.

Below is the results I get from running this code:
https://github.com/EnviroDIY/YosemitechModbus/tree/AnthonyTest/examples/GetValues_Y511_AltSoftSerial
`
Yosemitech Y511 sensor for Turbidity
Waiting for sensor and adapter to be ready.
Getting sensor version.
    Current Hardware Version: 1.02
    Current Software Version: 1.07
Getting sensor serial number.
    Serial Number: YL0917120505
Getting sensor calibration equation.
    Current Calibration Equation: final = 1.00*raw + 0.16
Getting sensor cleaning interval.
    Sensor auto-cleaning interval: 0 minutes
Resetting cleaning interval to 30 minutes.
    Set interval failed!
Starting sensor measurements
    Failed to start measuring!
Allowing sensor to stabilize..
10...9...8...7...6...5...4...3...2...1...

Activating brush.
    Failed to activate brush!
Continuing to stabilize..
12...11...10...9...8...7...6...5...4...3...2...1...

Temp(°C)  Turbidity(NTU)
21.44      0.00
21.44      0.00
21.46      0.00
`

SRGDamia1, 2018-01-24:
Are you definitely able to change the brushing rate using a computer or some other modbus controller [not using this library or SensorModbusMaster]?  If you can control the brush from a computer but not this library, it could be an issue with this library.  Otherwise, it is your sensors that are the problem.

aufdenkampe, 2018-01-24:
I just finished using the USB-to-RS485 converter and the Modbus Runner 1.9.1.3 desktop software (for Windows) that Yosemitech provided. I was able to get all my sensors to respond to all commands, including Set SlaveID and Activate Brush.

However, I am still not able to Set SlaveID with the following AltSoftSerial sketch, a 9V power source (https://www.pololu.com/product/2116) and the SCM TTL to RS485 Adapter 485 to Serial Port UART Level Converter Module 3.3V 5V (https://www.amazon.com/gp/product/B010723BCE/).
https://github.com/EnviroDIY/YosemitechModbus/tree/AnthonyTest/utilities/ChangeSlaveID_AltSoftSerial

This is my response.
My GetSlaveID sketch works fine.

```
ChangeSlaveID_AltSoftSerial.ino
Allowing sensor and adapter to warm up
10...9...8...7...6...5...4...3...2...1...

Changing Slave ID:
Address change failed!
```

SRGDamia1, 2018-01-31:
I'm sorry, I finally go a chance to run the sketch with my own Y511B sensor and found it didn't work.  I didn't think I' would actually publish something on GitHub knowing id didn't work, so I dug through my emails and found this from Zoe at Yosemitech:

> The software 1.06 is too old to respond modbus commands to control the brush .The brush just auto work every 30 minute. The Modbus manual says” This document is about MODBUS of optical Self-cleaning Turbidity probes with hardware Rev1.0 and software Rev1.7 or later” at the beginning. The turbidity you have is produced last year and we are constantly improving function.

aufdenkampe, 2018-01-31:
Here's another GetValues output from an older turbidity sensor. Note that extra character in the output form the Serial Number.

```
Yosemitech Y511 sensor for Turbidity
Waiting for sensor and adapter to be ready.
Getting sensor version.
    Current Hardware Version: 1.03
    Current Software Version: 2.01
Getting sensor serial number.
    Serial Number: )YL2917050209
Getting sensor calibration equation.
    Current Calibration Equation: final = 1.00*raw + 0.00
Getting sensor cleaning interval.
    Sensor auto-cleaning interval: 30 minutes
Resetting cleaning interval to 30 minutes.
    Set interval failed!
Starting sensor measurements
    Measurements started.
Allowing sensor to stabilize..
10...9...8...7...6...5...4...3...2...1...

Activating brush.
    Brush activated.
Continuing to stabilize..
12...11...10...9...8...7...6...5...4...3...2...1...

Temp(°C)  Turbidity(NTU)
23.50      -4.62
23.37      -4.81
23.50      -4.90
23.50      -4.96
23.50      -5.38
23.50      -5.40
```

Verbose debugging output:

```
⸮Yosemitech Y511 sensor for Turbidity
Waiting for sensor and adapter to be ready.
Getting sensor version.
Raw Request >>> {0x03, 0x03, 0x07, 0x00, 0x00, 0x02, 0xC4, 0x9D}
Raw Response (9 bytes) <<< {0x03, 0x03, 0x04, 0x01, 0x03, 0x02, 0x01, 0xE8, 0xAF}
    Current Hardware Version: 1.03
    Current Software Version: 2.01
Getting sensor serial number.
Raw Request >>> {0x03, 0x03, 0x09, 0x00, 0x00, 0x07, 0x06, 0x76}
Raw Response (19 bytes) <<< {0x03, 0x03, 0x0E, 0x29, 0x59, 0x4C, 0x32, 0x39, 0x31, 0x37, 0x30, 0x35, 0x30, 0x32, 0x30, 0x39, 0x00, 0xFC, 0x64}
    Serial Number: )YL2917050209
Getting sensor calibration equation.
Raw Request >>> {0x03, 0x03, 0x11, 0x00, 0x00, 0x04, 0x40, 0xD7}
Raw Response (13 bytes) <<< {0x03, 0x03, 0x08, 0x00, 0x00, 0x80, 0x3F, 0x00, 0x00, 0x00, 0x00, 0x95, 0xAA}
    Current Calibration Equation: final = 1.00*raw + 0.00
Getting sensor cleaning interval.
Raw Request >>> {0x03, 0x03, 0x32, 0x00, 0x00, 0x01, 0x8B, 0x50}
Raw Response (7 bytes) <<< {0x03, 0x03, 0x02, 0x1E, 0x00, 0xC8, 0x24}
    Sensor auto-cleaning interval: 30 minutes
Resetting cleaning interval to 30 minutes.
Raw Request >>> {0x03, 0x06, 0x32, 0x00, 0x1E, 0x00, 0x8F, 0x30}
No response received.
Raw Request >>> {0x03, 0x06, 0x32, 0x00, 0x1E, 0x00, 0x8F, 0x30}
No response received.
Raw Request >>> {0x03, 0x06, 0x32, 0x00, 0x1E, 0x00, 0x8F, 0x30}
No response received.
Raw Request >>> {0x03, 0x06, 0x32, 0x00, 0x1E, 0x00, 0x8F, 0x30}
No response received.
Raw Request >>> {0x03, 0x06, 0x32, 0x00, 0x1E, 0x00, 0x8F, 0x30}
No response received.
Raw Request >>> {0x03, 0x06, 0x32, 0x00, 0x1E, 0x00, 0x8F, 0x30}
No response received.
Raw Request >>> {0x03, 0x06, 0x32, 0x00, 0x1E, 0x00, 0x8F, 0x30}
No response received.
Raw Request >>> {0x03, 0x06, 0x32, 0x00, 0x1E, 0x00, 0x8F, 0x30}
No response received.
Raw Request >>> {0x03, 0x06, 0x32, 0x00, 0x1E, 0x00, 0x8F, 0x30}
No response received.
Raw Request >>> {0x03, 0x06, 0x32, 0x00, 0x1E, 0x00, 0x8F, 0x30}
No response received.
    Set interval failed!
Starting sensor measurements
Raw Request >>> {0x03, 0x03, 0x25, 0x00, 0x00, 0x00, 0x4F, 0x24}
Raw Response (5 bytes) <<< {0x03, 0x03, 0x00, 0x81, 0x30}
    Measurements started.
Allowing sensor to stabilize..
10...9...8...7...6...5...4...3...2...1...

Activating brush.
Raw Request >>> {0x03, 0x10, 0x31, 0x00, 0x00, 0x00, 0x00, 0x57, 0x54}
Raw Response (8 bytes) <<< {0x03, 0x10, 0x31, 0x00, 0x00, 0x00, 0xCF, 0x17}
    Brush activated.
Continuing to stabilize..
12...11...10...9...8...7...6...5...4...3...2...1...

Temp(°C)  Turbidity(NTU)
Raw Request >>> {0x03, 0x03, 0x26, 0x00, 0x00, 0x04, 0x4E, 0xA3}
Raw Response (13 bytes) <<< {0x03, 0x03, 0x08, 0x00, 0x00, 0xBD, 0x41, 0x49, 0xC7, 0x94, 0xC0, 0x70, 0x7C}
23.62      -4.65
Raw Request >>> {0x03, 0x03, 0x26, 0x00, 0x00, 0x04, 0x4E, 0xA3}
Raw Response (13 bytes) <<< {0x03, 0x03, 0x08, 0x00, 0x00, 0xBD, 0x41, 0x0D, 0x09, 0x9C, 0xC0, 0x02, 0xB3}
23.62      -4.88
Raw Request >>> {0x03, 0x03, 0x26, 0x00, 0x00, 0x04, 0x4E, 0xA3}
Raw Response (13 bytes) <<< {0x03, 0x03, 0x08, 0x00, 0x00, 0xBD, 0x41, 0x94, 0x51, 0xA0, 0xC0, 0xBC, 0xFC}
23.62      -5.01
```

aufdenkampe, 2018-01-31:
Here's output from the same sensor when running commands using YosemiTech's Modbus Runner v1.9.1.3 program:

GetSlaveID: (Answer = 03)
```
SEND: FF 03 30 00 00 01 9E D4 
GET: FF 03 
GET: 02 03 
GET: 00 
GET: 91 60 
```

SetSlaveID: (Answer = 01)
```
SEND: 03 10 30 00 00 01 02 01 00 8E A3 
GET: 03 10 30 00 00 01 0F 2B 
```

aufdenkampe, 2018-01-31:
Here's the results from the ChangeSlaveID_AltSoftSerial.ino, with debugging on.

```
ChangeSlaveID_AltSoftSerial.ino
Allowing sensor and adapter to warm up
10...9...8...7...6...5...4...3...2...1...

Changing Slave ID:
Raw Request >>> {0x03, 0x06, 0x30, 0x00, 0x01, 0x00, 0x86, 0xB8}
No response received.
Raw Request >>> {0x03, 0x06, 0x30, 0x00, 0x01, 0x00, 0x86, 0xB8}
No response received.
Raw Request >>> {0x03, 0x06, 0x30, 0x00, 0x01, 0x00, 0x86, 0xB8}
No response received.
Raw Request >>> {0x03, 0x06, 0x30, 0x00, 0x01, 0x00, 0x86, 0xB8}
No response received.
Raw Request >>> {0x03, 0x06, 0x30, 0x00, 0x01, 0x00, 0x86, 0xB8}
No response received.
Raw Request >>> {0x03, 0x06, 0x30, 0x00, 0x01, 0x00, 0x86, 0xB8}
No response received.
Raw Request >>> {0x03, 0x06, 0x30, 0x00, 0x01, 0x00, 0x86, 0xB8}
No response received.
Raw Request >>> {0x03, 0x06, 0x30, 0x00, 0x01, 0x00, 0x86, 0xB8}
No response received.
Raw Request >>> {0x03, 0x06, 0x30, 0x00, 0x01, 0x00, 0x86, 0xB8}
No response received.
Raw Request >>> {0x03, 0x06, 0x30, 0x00, 0x01, 0x00, 0x86, 0xB8}
No response received.
Address change failed!
```

aufdenkampe, 2018-01-31:
Here's output from the same sensor when running other commands using YosemiTech's *Modbus Runner v1.9.1.3* program:

GetSN (Answer = YL2917050209)
```
SEND: 03 03 09 00 00 07 06 76 
GET: 03 03 0E 29 59 4C 32 39 31 37 30 35 30 32 30 39 00 FC 64 
```

GetVersion (Answer = hw 1.3,sw 2.1)
```
SEND: 03 03 07 00 00 02 C4 9D 
GET: 03 03 04 01 03 02 01 E8 AF 
```

GetKB (user coefficients, answers K=1.000000, B=0.000000)
```
SEND: 03 03 11 00 00 04 40 D7 
GET: 03 03 08 00 00 80 3F 00 00 00 00 95 AA 
```

SetKB (user coefficients, answers K=1.000000, B=0.000000)
```
SEND: 03 10 11 00 00 04 08 00 00 80 3F 00 00 00 00 03 AF 
GET: 03 10 11 00 00 04 C5 14 
```

ActivateBrush
```
SEND: 03 10 31 00 00 00 00 57 54 
GET: 03 10 31 00 00 00 CF 17 
```

GetBrushInterval (Answer = 30)
```
SEND: 03 03 32 00 00 01 8B 50 
GET: 03 03 02 1E 00 C8 24 
```

SetBrushInterval (set to = 60)
```
SEND: 03 10 32 00 00 01 02 3C 00 BD F3 
GET: 03 10 32 00 00 01 0E 93 
```

SetBrushInterval (set to = 30)
```
SEND: 03 10 32 00 00 01 02 1E 00 A5 53 
GET: 03 10 32 00 00 01 0E 93 
```

Activate Sensor (for DO+Tur+Chl+COD)
```
SEND: 03 03 25 00 00 00 4F 24 
GET: 03 03 00 81 30 
```

GetValues  (for DO+Tur+Chl+COD), 5 times:
```
SEND: 03 03 26 00 00 04 4E A3 
GET: 03 03 08 00 00 C3 41 9C 60 B5 C0 EB DD 
SEND: 03 03 26 00 00 04 4E A3 
GET: 03 03 08 00 00 C3 41 9C 60 B5 C0 EB DD 
SEND: 03 03 26 00 00 04 4E A3 
GET: 03 03 08 00 00 C3 41 89 D7 B5 C0 5F F7 
SEND: 03 03 26 00 00 04 4E A3 
GET: 03 03 08 00 00 C3 41 89 D7 B5 C0 5F F7 
SEND: 03 03 26 00 00 04 4E A3 
GET: 03 03 08 00 00 C3 41 7D 3B B5 C0 AC 32 
```
Answers
```
-5.668043	24.375000
-5.668043	24.375000
-5.682560	24.375000
-5.682560	24.375000
-5.663512	24.375000
```

SRGDamia1, 2018-01-31:
I think I've fixed things now.   Please try updating SensorModbusMaster and this and running again.

aufdenkampe, 2018-01-31:
Will do! Thank you! I'll let you know what I find!

SRGDamia1, 2018-01-31:
Also, please check and see if it fixes #2

aufdenkampe, 2018-01-31:
Bummer! I didn't work. My output is identical to what I got above (see https://github.com/EnviroDIY/YosemitechModbus/issues/3#issuecomment-362012539).

In other words, it is still sending the 0x06 command to write to a single register.
Did you get it to work on your machine? I double checked that I'm compiling with your new code.

SRGDamia1, 2018-01-31:
Did you update SensorModbusMaster?  Can you check the version numbers of both?

SRGDamia1, 2018-01-31:
Are you using PlatformIO or the Arduino IDE?  If PlatformIO, check if you have both libraries installed by git or by library name/number.  If they are installed by library name/number in PlatformIO, the web crawler that updates its indices only runs every day or so so you won't catch the update until it's run.  Make sure you don't have the same library twice.

[image]

I'll update the zip in the libraries directory.

aufdenkampe, 2018-01-31:
It worked!

```
ChangeSlaveID_AltSoftSerial.ino
Allowing sensor and adapter to warm up
10...9...8...7...6...5...4...3...2...1...

Changing Slave ID:
Raw Request >>> {0x03, 0x10, 0x30, 0x00, 0x00, 0x01, 0x02, 0x01, 0x00, 0x8E, 0xA3}
Raw Response (8 bytes) <<< {0x03, 0x10, 0x30, 0x00, 0x00, 0x01, 0x0F, 0x2B}
Address change complete.
```

It was my mistake. I had properly updated [SensorModbusMaster](https://github.com/EnviroDIY/SensorModbusMaster) to v0.6.2 but didn't notice that you also made changes to [YosemitechModbus](https://github.com/EnviroDIY/YosemitechModbus). Updating that to Master did the trick!

aufdenkampe, 2018-01-31:
Here are the results (with debugging output) from *GetValues_Y511_AltSoftSerial.ino*.
In short, 2 bugs were fixed compared to my [output above](https://github.com/EnviroDIY/YosemitechModbus/issues/3#issuecomment-362008862):
* the serial number was printed without the extra leading ")" character
* the cleaning interval was successfully reset
However, the Activate Brush command did not get the brush to spin, as  I can do using the Modbus Runner v1.9.1.3 program.

```
Yosemitech Y511 sensor for Turbidity
Waiting for sensor and adapter to be ready.
Getting sensor version.
Raw Request >>> {0x03, 0x03, 0x07, 0x00, 0x00, 0x02, 0xC4, 0x9D}
Raw Response (9 bytes) <<< {0x03, 0x03, 0x04, 0x01, 0x03, 0x02, 0x01, 0xE8, 0xAF}
    Current Hardware Version: 1.03
    Current Software Version: 2.01
Getting sensor serial number.
Raw Request >>> {0x03, 0x03, 0x09, 0x00, 0x00, 0x07, 0x06, 0x76}
Raw Response (19 bytes) <<< {0x03, 0x03, 0x0E, 0x29, 0x59, 0x4C, 0x32, 0x39, 0x31, 0x37, 0x30, 0x35, 0x30, 0x32, 0x30, 0x39, 0x00, 0xFC, 0x64}
    Serial Number: YL2917050209
Getting sensor calibration equation.
Raw Request >>> {0x03, 0x03, 0x11, 0x00, 0x00, 0x04, 0x40, 0xD7}
Raw Response (13 bytes) <<< {0x03, 0x03, 0x08, 0x00, 0x00, 0x80, 0x3F, 0x00, 0x00, 0x00, 0x00, 0x95, 0xAA}
    Current Calibration Equation: final = 1.00*raw + 0.00
Getting sensor cleaning interval.
Raw Request >>> {0x03, 0x03, 0x32, 0x00, 0x00, 0x01, 0x8B, 0x50}
Raw Response (7 bytes) <<< {0x03, 0x03, 0x02, 0x1E, 0x00, 0xC8, 0x24}
    Sensor auto-cleaning interval: 30 minutes
Resetting cleaning interval to 30 minutes.
Raw Request >>> {0x03, 0x10, 0x32, 0x00, 0x00, 0x01, 0x02, 0x1E, 0x00, 0xA5, 0x53}
Raw Response (8 bytes) <<< {0x03, 0x10, 0x32, 0x00, 0x00, 0x01, 0x0E, 0x93}
    Reset.
Starting sensor measurements
Raw Request >>> {0x03, 0x03, 0x25, 0x00, 0x00, 0x00, 0x4F, 0x24}
Raw Response (5 bytes) <<< {0x03, 0x03, 0x00, 0x81, 0x30}
    Measurements started.
Allowing sensor to stabilize..
10...9...8...7...6...5...4...3...2...1...

Activating brush.
Raw Request >>> {0x03, 0x10, 0x31, 0x00, 0x00, 0x00, 0x00, 0x57, 0x54}
Raw Response (8 bytes) <<< {0x03, 0x10, 0x31, 0x00, 0x00, 0x00, 0xCF, 0x17}
    Brush activated.
Continuing to stabilize..
12...11...10...9...8...7...6...5...4...3...2...1...

Temp(°C)  Turbidity(NTU)
Raw Request >>> {0x03, 0x03, 0x26, 0x00, 0x00, 0x04, 0x4E, 0xA3}
Raw Response (13 bytes) <<< {0x03, 0x03, 0x08, 0x00, 0x00, 0xB7, 0x41, 0xA9, 0xC3, 0x98, 0xC0, 0x03, 0xD7}
22.87      -4.77
Raw Request >>> {0x03, 0x03, 0x26, 0x00, 0x00, 0x04, 0x4E, 0xA3}
Raw Response (13 bytes) <<< {0x03, 0x03, 0x08, 0x00, 0x00, 0xB7, 0x41, 0xA3, 0xC8, 0x9D, 0xC0, 0x72, 0x9D}
22.87      -4.93
Raw Request >>> {0x03, 0x03, 0x26, 0x00, 0x00, 0x04, 0x4E, 0xA3}
Raw Response (13 bytes) <<< {0x03, 0x03, 0x08, 0x00, 0x00, 0xB7, 0x41, 0xF9, 0x0D, 0xA1, 0xC0, 0x61, 0x78}
22.87      -5.03
Raw Request >>> {0x03, 0x03, 0x26, 0x00, 0x00, 0x04, 0x4E, 0xA3}
Raw Response (13 bytes) <<< {0x03, 0x03, 0x08, 0x00, 0x00, 0xB7, 0x41, 0x03, 0xAF, 0xB1, 0xC0, 0xFD, 0x82}
22.87      -5.55
```

aufdenkampe, 2018-01-31:
I don't understand why the brush doesn't activate. The Send request and the Get response are identical between the *GetValues_Y511_AltSoftSerial.ino* and the *Modbus Runner v1.9.1.3* program

The manual does specify a 20s delay after activating the brush, before starting measurements, but it seems I can do both simultaneously using the *Modbus Runner v1.9.1.3* program, so that's not a hardware or firmware constraint.

BTW, the *GetValues_Y511_AltSoftSerial.ino* also fails to produce a Get or Set Brush Interval command. I can change these with the *Modbus Runner v1.9.1.3* program, but *GetValues_Y511_AltSoftSerial.ino*  actually seems to reset the brush interval to 30 minutes before reading the actual interval (i.e. I set the brush interval to 3 minutes with ModbusRunner, but my INO still reads 30, then I check with ModbusRunner and it now reads 30, even when the INO was trying to set it to 3 minutes).

BTW2, I'm still doing all my compiling and uploading using the Arduino IDE.

SRGDamia1, 2018-01-31:
Really?  It didn't activate?  The command sent to the sensor was identical and it responded as if it did in the output you posted.

From what you posted with the modbus Runner:
> ActivateBrush
> SEND: 03 10 31 00 00 00 00 57 54 
> GET: 03 10 31 00 00 00 CF 17 

From your output above with the Mayfly:

> Activating brush.
> Raw Request >>> {0x03, 0x10, 0x31, 0x00, 0x00, 0x00, 0x00, 0x57, 0x54}
> Raw Response (8 bytes) <<< {0x03, 0x10, 0x31, 0x00, 0x00, 0x00, 0xCF, 0x17}

SRGDamia1, 2018-02-01:
I'm sorry.  I'm kind of lost.  From what I can see in what you've posted, this Yosemitech library is sending an identical command as Modbus Runner and the Modbus response from the sensor is also identical.  Can you try creating a sketch just to test the brush activation.  Maybe it has something to do with how long it has been since the brush was previously activated..

aufdenkampe, 2018-02-01:
Yup. Thats right that the "Yosemitech library is sending an identical command as Modbus Runner and the Modbus response from the sensor is also identical." However, the brush doesn't actually spin when we use the YosemiTech library, but it does when we use Modbus Runner.  It's very perplexing.

I'll try your suggestion, to create a sketch just to test the brush. However, the respond to the Activate Brush command from *Modbus Runner* is immediate and never fails, so I'm suspecting something else is going on. One thing to note is that the USB-RS485 adapter that I use with *Modbus Runner* supplies 5V power, while I've set up the TTL-RS485 converter with a [9V step-up boost](https://www.pololu.com/product/2116)(which was recommended by @s-hicks2) from the Mayfly's 5V line. Its still perplexing to me that 5V always works from the USB-RS485 adapter, whereas I needed to boost to >6V to get the sensors with brushes to respond from the Mayfly (see https://github.com/EnviroDIY/YosemitechModbus/issues/3#issuecomment-359947426).

aufdenkampe, 2018-02-02:
I reached out last night for help & ideas from YosemiTech with this email:

> We have the following sensors with brushes
> * ​Y511 Turbidity software versions 1.6, 1.9, 2.1 and 2.3
> * Y514 Chlorophyl software v2.0
>
> For all of them, we notice that the first time we power them, the brush automatically activates.
> However, for the more recent versions (everything but our oldest, Y511 v1.6), after we attempted to reset the brush interval and trigger activate for the brush, we find that the brush no longer automatically activates once power gets turned on. Is there a reason for this? How can we get it automatically activate the brush when power comes on?
>...
> We notice using Modbus Runner and your USB-RS485 converter, that when we turn off the power to the sensor, that the Brush interval seems to get reset to 30 minutes, regardless of what it was set to when power was last on. Is this expected behavior?

I got this response from Wenxing Gao <wenxing@yosemitech.com>:
  >   Our new version of all sensors with wiper brush cannot automatically activate brush once power gets turned on . And also cannot remember the Brush interval time that was set when power last on.
>      The code seems OK. The reason wiper do not work maybe is the power offered too low to make the wiper activate. Please make sure  the power of sensor is 12V ,50mA.

aufdenkampe, 2018-02-02:
Success! I am able to manually Activate the Brush, using the same code I've been using (https://github.com/EnviroDIY/YosemitechModbus/tree/AnthonyTest/examples/GetValues_Y511_AltSoftSerial).

The solution *did not* require increasing the voltage, as suggested by Wenxing Gao (https://github.com/EnviroDIY/YosemitechModbus/issues/3#issuecomment-362643084).

The solution was to change my interface hardware. I switched:
* from using a prototype board using [SCM TTL-RS485 converter](https://www.amazon.com/gp/product/B010723BCE/) with a [Pololu 9V step-up boost](https://www.pololu.com/product/2116) 
* to using a wiring harness using the same model  [SCM TTL-RS485 converter](https://www.amazon.com/gp/product/B010723BCE/) but with a [Boost Converter Module XL6009 DC to DC 3.0-30 V to 5-35 V Output Voltage Adjustable Step-up Circuit Board ](https://www.amazon.com/gp/product/B06XWSV89D/).

The wiring harness worked for triggering Activate brush for any voltage from 5.0V to 12.0V!

So, the question for me to figure out is whether:
* the [Pololu 9V step-up boost](https://www.pololu.com/product/2116) isn't an appropriate component (all indications are that it is, because it should be able to handle >500 mA); or
* I had a faulty [SCM TTL-RS485 converter](https://www.amazon.com/gp/product/B010723BCE/) in my first setup.
* I had faulty wiring in my first setup.

SRGDamia1, 2018-02-02:
Great!  I'm going to close this issue then.

I actually just fried one of my RS485 adapters...

closed 2018-02-02 by SRGDamia1

aufdenkampe, 2018-02-02:
Yikes, how did you fry your RS485 adapter? Which one did you fry? It would be good to know what voltages these various devices can handle.

SRGDamia1, 2018-02-02:
I think it was one of the SCM boards.. I sort-of fried the Mayfly it was attached to, too.  I accidentally plugged the pin that should have been attached to ground to 5V while the 3.3V was already plugged into the adapter.  Everything got very hot and started responding very weirdly.  I'm just really glad I didn't also fry my sensors.

aufdenkampe, 2018-02-05:
Wenxing Gao <wenxing@yosemitech.com> now has a GitHub account.
Thanks @ZoeWXGao!