EnviroDIY/YosemitechModbus #6: Add support for Y4000 Multi-Parameter Sonde

aufdenkampe, 2018-02-01:
I need to get this working by mid-March.
Here's the Modbus Manual: 
https://github.com/EnviroDIY/YosemitechModbus/blob/master/doc/Y4000-Sonde-1.6-ModbusInstruction-en.pdf

It appears to have very similar functions, but GetValues returns floats for all possible probes in a single response, returning 0 if a probe of a given type is not connected.

It's confusing how GetSN works, as it states that "Each sensor probe has a unique SN.", but the example response frame one has one shown. It appears the the SlaveID is for the sonde, not for the different probes connected to the sonde.

There are new set of functions, including:
* get/set real clock time
* Get supply voltage

SRGDamia1, 2018-02-01:
Unfortunately, I don't have one of these to test and will have very little time to help with this unless you convince Arscott otherwise.

aufdenkampe, 2018-04-25:
Largely done with PR #12, but still need:
* some utility functions have not yet been completed for Y4000
* Better documentation for Y4000

aufdenkampe, 2020-10-02:
Closing this, as we mostly figured it out and also still have issue #13 open.

closed 2020-10-02 by aufdenkampe