EnviroDIY/YosemitechModbus #9: Temp and Param switched in Example/GetValues

aufdenkampe, 2018-02-07:
@SRGDamia1, I know you made a fix with https://github.com/EnviroDIY/YosemitechModbus/commit/3f8eef2bc3d2f30ae628ae6b68b68f9e86d5d962, and I noted that I had sometimes seen Temp and Param switched in examples.

However, now when I run https://github.com/EnviroDIY/YosemitechModbus/tree/master/examples/GetValues, the two are switched all the time.

They are correct, however, in the latest code using ModularSensors.

Ideas?

aufdenkampe, 2018-02-07:
Here is a result for turbidity. I saw the same for Conductivity.

```
Yosemitech Y511 sensor for Turbidity
Waiting for sensor and adapter to be ready.
Getting sensor version.
    Current Hardware Version: 1.03
    Current Software Version: 2.03
Getting sensor serial number.
    Serial Number: YL2917113008
Getting sensor calibration equation.
    Current Calibration Equation: final = 1.00*raw + 0.00
Getting sensor cleaning interval.
    Sensor auto-cleaning interval: 30 minutes
Resetting cleaning interval to 30 minutes.
    Reset.
Starting sensor measurements
    Measurements started.
Allowing sensor to stabilize..
10...9...8...7...6...5...4...3...2...1...

Activating brush.
    Brush activated.
Continuing to stabilize..
12...11...10...9...8...7...6...5...4...3...2...1...

Temp(°C)  Turbidity(NTU)
-0.15      22.25
-0.15      22.25
-0.15      22.25
-0.15      22.25
```

SRGDamia1, 2018-02-08:
It was just that the variable order was wrong in the getValues command in the example.  Fixed by https://github.com/EnviroDIY/YosemitechModbus/commit/d6d11511ec4a355f5e35fbd8cee54b18ea8f07b6.

closed 2018-02-08 by SRGDamia1