EnviroDIY/YosemitechModbus #10: Optimize approach for combined voltage boost & RS-485 conversion

aufdenkampe, 2018-03-24:
Let's use this as a place to document our work with @bschulz1701.

**Goal** is to have a structurally stable wiring harness or shield, that is easy to assemble and that:
* Utilize the [WINGONEER TTL to RS485 Adapter / 485 to Serial Port UART Level Converter Module (XY-017)](https://www.amazon.com/gp/product/B06XHH6B6R) via the [AltSoftSerial](https://github.com/PaulStoffregen/AltSoftSerial) library, which requires Mayfly pins D5 & D6 for RX/TX and Mayfly's switched 3.3V power
* Power the Modbus sensor off the Mayfly switched 5V power which we then boost to 9V (most Yosemitech sensors) or to 12V (for the Yosemitech sondes).

**Background**:
* We used to use the wiring approach shown on the main readme here, https://github.com/EnviroDIY/YosemitechModbus#suggested-setup-with-an-envirodiy-mayfly, but we found it to be unstable because it used SoftwareSerial rather than AltSoftSerial. See #3 starting here: https://github.com/EnviroDIY/YosemitechModbus/issues/3#issuecomment-359947426
* Working on Sonde support in issue #6

aufdenkampe, 2018-03-26:
@bschulz1701, you can test the turbidity sensor I left for you by:
* first running this sketch to get slave ID: https://github.com/EnviroDIY/YosemitechModbus/tree/master/utilities/GetSlaveID
* then run this sketch, after changing to the slaveID: https://github.com/EnviroDIY/YosemitechModbus/tree/master/examples/GetValues

aufdenkampe, 2018-03-26:
Here's the schematic for that shield that Beth and I developed, using the [Pololu 9V Step-Up Voltage Regulator U3V12F9](https://www.pololu.com/product/2116), which worked great for everything except powering the brush:

[image]

SRGDamia1, 2018-03-26:
Yeah.. My wish-list definitely includes a single board that includes both the RS485 conversion and the power boost, preferably with grove on one end and multiple sets of A/B/Gnd screw terminals on the other.

SRGDamia1, 2018-03-26:
I also want a nice little micro-shield like the ones we have for a grove connection on Serial/Serial1 but for AltSoftSerial.

aufdenkampe, 2018-03-26:
@SRGDamia1, These are photos of the ones Beth and I developed one as a prototype (schematic above), for AltSoftSerial.
[image]
[image]
[image]

They work great for everything but the brush function. They are also a total pain to solder all those jumper wires.

@bschulz1701 is working on helping us find a setup that works for all functions, and then print out a circuit board, so that we just have to solder the components. We'll share the Eagle files in this repo.

aufdenkampe, 2018-03-28:
@bschulz1701, Thanks for asking which repo to put your hardware work. At the moment our hardware info @is at https://github.com/EnviroDIY/YosemitechModbus#hardware

However, given that we want to use this for all Modbus devices, I think it might make sense to put it here: https://github.com/EnviroDIY/SensorModbusMaster

Let's do that. I just created a "Hardware" branch in SensorModbusMaster. In that branch is a "Hardware" folder. Let's use that to put all the hardware info, eventually including instructions in that new readme file https://github.com/EnviroDIY/SensorModbusMaster/tree/hardware

closed 2018-05-08 by SRGDamia1