EnviroDIY/YosemitechModbus #17: Does not work with esp8266 version >=2.6.0

crash007, 2020-01-22:
Hi,
I dont know if this is a yosemite issue or more with esp8266.
I have a older wemos v3 esp8266 and this library together with sensorModMaster does not work with newer versions than 2.5.2.
According to https://github.com/esp8266/Arduino/releases something happens to serial in 2.6.0 which is then fixed in 2.6.1 but when running I dont get any readings on the bus.
So to get any readings im forced to use 2.5.
Maybe you guys knows anything about this?

SRGDamia1, 2020-01-22:
I don't follow the ESP-Arduino community, but nothing in their change log looks like it should have made a difference to this library.  Are you using software serial?  Are you sure you've ordered the arguments of the constructor and the begin functions correctly?  What version of SensorModbusMaster and of this library are you using?  There was a bug in SensorModbusMaster related to the debugging functionality that was fixed with 0.6.8 of that library.

This library is all high level functionality - it should work on any Arduino style core that implements a stream object.

crash007, 2020-01-23:
Im using Master from yesterday.
This library works fine if I use esp8266 release version 2.5.2 or older. Starting with 2.6.0 they implement a breaking change which is later fixed in 2.6.1. But if i try run with the latest version 2.6.3 and 2.6.1 I cant get any readings from the bus.
If this library is high level functionality i think this issue might be with esp8266 instead. 
Looking at  https://github.com/esp8266/Arduino/releases can you see what might have changed so its not possible to read the bus any longer starting with 2.6.0?
Im using a older wemos 8266 v3

SRGDamia1, 2020-01-23:
Are you using software or hardware serial?  Do you have debugging turned on?  (ie, ```sensor.setDebugStream(&Serial);```

SRGDamia1, 2020-01-23:
Does it make any difference if you call the modbus commands directly using SensorModbusMaster instead of using the YosemitechModbus library?  

Try getting a slave address:
```cpp
    // Start the modbus instance
    modbus.begin(modbusAddress, modbusSerial, DEREPin);
    byte command[8] = {0xFF, 0x03, 0x30, 0x00, 0x00, 0x01, 0x9E, 0xD4};
    int respSize = modbus.sendCommand(command, 8);
```

crash007, 2020-01-24:
Its no difference with or without debug. I will try getting a slave address this weekend.
Looking at issues with espsoftserial related to 2.6.0 saw this:

> ESP8266 core 2.5.0 referenced a version of EspSoftwareSerial that's a completely different beast. I doubt it works on the latest ESP8266 core release, due to changed timings in the core BSP.
The internal buffers in EspSoftwareSerial should not be used to buffer large datagrams, please read as often as possible into user provided buffers, and reduce the size of the SW serial buffer accordingly.

Correct me if im wrong I think esp8266 2.5.0 uses espsoftserial  3.4.1 while esp8266 2.6.3 uses 6.4.0

crash007, 2020-01-25:
I ran the getSlaveId example which doesnt use any modbus libraries.
Output running with esp8266 board version 2.5.2

> Scanning for Yosemitech modbus sensors....
> ------------------------------------------
> Modbus Address ------ Sensor Serial Number
> {0x01, 0x03, 0x09, 0x00, 0x00, 0x07, 0x07, 0x94}
>      0x01      ------    YL0119080929LC @⸮⸮⸮?⸮⸮⸮?⸮⸮⸮?⸮ @

Running with board version 2.6.3:

>  Modbus Address ------ Sensor Serial Number
> {0x01, 0x03, 0x09, 0x00, 0x00, 0x07, 0x07, 0x94}
>      0x01      ------    ????????????
> {0x02, 0x03, 0x09, 0x00, 0x00, 0x07, 0x07, 0xA7}
>      0x02      ------    ????????????
> {0x03, 0x03, 0x09, 0x00, 0x00, 0x07, 0x06, 0x76}

SRGDamia1, 2020-01-25:
I'm sorry, I did some testing yesterday and should have updated you.  

The problem was that the readBytes function (used internally by SensorModbusMaster) was completely non-blocking in EspSoftwareSerial 6.4.0.  Because the ESP is very fast relative to the slow 9600 baud of the Yosemitech, it would think the response timed out immediately.  The developers of EspSoftwareSerial partially fixed the issue between when the ESP core was released and now and immediately responded and fixed the remaining timeout issue when I raised an issue with them yesterday.  I have not had a chance to test it yet, but I believe if you update EspSoftwareSerial to the brand-new 6.7.1 release it should work for you:  https://github.com/plerup/espsoftwareserial/releases

Let me know if that doesn't work.

crash007, 2020-01-25:
Thanks for your response @SRGDamia1. I just tried it with 6.7.1 but it did not work :/

SRGDamia1, 2020-01-25:
I'm sorry. I'll do more testing on Monday.

crash007, 2020-01-25:
Appriciate your help. Have great weekend

SRGDamia1, 2020-01-27:
Good and bad news:  It works for me; I don't know why it wouldn't work for you.

Board:  Adafruit Feather Huzzah (ESP12S=ESP8266)
ESP Core: 2.6.3
ESPSoftwareSerial: 6.7.1
SensorModbusMaster: 0.6.8
YosemitechModbus: 0.2.2

Yosemitech Y532 (pH sensor)

Running "getValues" example with these changes:
- Include SoftwareSerial.h, not AltSoftSerial.h
- Corrected modbus model and address to the ones I happened to have on hand
- using these pins:
```cpp
const int DEREPin = 14;
const int SSRxPin = 13;
const int SSTxPin = 12;
```
- deleted lines 56-59 (sensor power and adapter power pins) and had sensor on separate continuous power supply

I tested with two different RS485 adapters, one with automatic flow control and another with manual flow control.  Both of the RS485 adapters I tried were 3.3V logic level - no level shifters needed - and both worked fine.  Having debugging printing on or off made no difference; it worked either way.

crash007, 2020-02-04:
I tried again with arduino 1.8.10 maybe i, doing something wrong selecting the newer version of espsoftserial. The sensor i been working with is a Y504.