EnviroDIY/YosemitechModbus #19: Y533 and Y532 Deviation In Measurements

GArrigotti-cws, 2021-06-24:
Rather than e-mail back and forth, after using the example software "Get Values" I noticed that the output for the Y533 Oxygen Reduction Potential sensor was providing the mV as pH, not actually mV.

```terminal
Yosemitech Y533 sensor for ORP
Waiting for sensor and adapter to be ready.
Getting sensor version.
    Current Hardware Version: 1.03
    Current Software Version: 2.03
Getting sensor serial number.     
    Serial Number: DYL4421032604  
Getting sensor calibration status.
    Status: 0x00
For Y4000, use YosemiTech software to get calibration parameters.
Starting sensor measurements
    Failed to start measuring!
Allowing sensor to stabilize..
10...9...8...7...6...5...4...3...2...1...

Temp(°C)  ORP(mV)
22.63      4.29
22.64      4.29
22.64      4.28
```

I traced the YosmitechModbus.cpp to yosmitech::getValues method with a switch statement.  I separated the Y532 pH and the Y533 statement out a bit, altering the Y532 start register and numeric quantity based on documentation to the following:

```cpp
        // Y532 (pH) or Y533 (ORP)
        case Y532:
        {
            if (modbus.getRegisters(0x03, 0x2800, 2))
            {
                parmValue = modbus.float32FromFrame(littleEndian, 3);
                tempValue = modbus.float32FromRegister(0x03,  0x2400, littleEndian);
                thirdValue = modbus.float32FromRegister(0x03,  0x1200, littleEndian);
                errorCode = 0x00;  // No errors
                return true;
            }
            break;
        }

        case Y533:
        {
            if (modbus.getRegisters(0x03, 0x2600, 4))
            {
                parmValue = modbus.float32FromFrame(littleEndian, 3);
                tempValue = modbus.float32FromRegister(0x03,  0x2400, littleEndian);
                thirdValue = modbus.float32FromRegister(0x03,  0x1200, littleEndian);
                errorCode = 0x00;  // No errors
                return true;
            }
            break;
        }
```

I now get the correct values from the sensor in the Get Values sketch.  I'm assuming we have to incorporate that change to potentially solve my issue?

aufdenkampe, 2021-07-01:
@GArrigotti-cws, thanks for your initial sleuthing of this issue!

I created a new `develop` branch, where I cleaned up the code for the Y532 pH sensor and added a variant of your code for the Y533 ORP sensor, fully separating their functions. 

I also looked back at my commit 574a31ce8a3755ce501438fd46fbc67808124c1b from October 2019, replicating that fix for ORP, which will be necessary to get all this to work in ModularSensors.

I'll close this issue and merge to master once everything is working with ModularSensors.

GArrigotti-cws, 2021-07-01:
@aufdenkampe Thank you, let me know when I can pull and begin for implementation.

closed 2021-07-01 by aufdenkampe

aufdenkampe, 2021-07-02:
@GArrigotti-cws, I finally now have ORP working in ModularSensors (using the [y533_ORP](https://github.com/EnviroDIY/ModularSensors/tree/y533_ORP) branch.

In a moment I'll commit an example into our private deployments repo.

GArrigotti-cws, 2021-07-02:
@aufdenkampe I do have uncommitted code for Fernhill, so be cautious with that particular deployment sketch.

aufdenkampe, 2021-07-02:
@GArrigotti-cws, I just updated your working sketch (Fernhill-1). Try that.

aufdenkampe, 2021-07-06:
@GArrigotti-cws, btw, I created a `[y533_ORP_backup](https://github.com/EnviroDIY/ModularSensors/tree/y533_ORP_backup) branch to archive my commits from Friday morning separately from the merged commits that Sara made later that afternoon.

So for your Fernhill deployments, I would recommend:
```ini
lib_deps =
    https://github.com/EnviroDIY/ModularSensors.git#y533_ORP_backup
```

GArrigotti-cws, 2021-07-06:
@aufdenkampe You added a backup label to your branch?

aufdenkampe, 2021-07-06:
I created a new branch, based on my last commit on Friday. You can see in this branch listing here (https://github.com/EnviroDIY/ModularSensors/branches) that my original branch has 47 more commits it in as the results of Sara's merge. Those commits are why you had a different experience compiling than I did just a few hours earlier.

aufdenkampe, 2021-07-06:
#include <LoggerBase.h>

aufdenkampe, 2021-07-08:
These fixes have now been released into ModularSensors v0.30.0.

https://github.com/EnviroDIY/ModularSensors/releases/tag/v0.30.0

commit 215f6302 (2021-07-01): Working with Y532 ORP!

Fixes #19
cc: @GArrigotti-cws