EnviroDIY/YosemitechModbus #23: Y4000 Not sending correct responses

TzionCastillo1, 2022-02-16:
Hi,

I am attempting to connect a Y4000 to my Mayfly using a RS485-TTL adapter with automatic flow control. Using this library's readWriteRegister example sketch, I am getting the following output on the Serial Monitor. It seems the sensor is sending its responses in a way that does not comply with their manual(see screenshot below) . Any help with this is appreciated.
[image]

Tzion

aufdenkampe, 2022-02-16:
Have you confirmed that the Modbus Address is correctly set in the sketch?

How did you Get or Set the Modbus address on the Y4000? The software from YosemiTech uses integer data types, but the sketches in this library all use the hexadecimal byte representation. These are only similar for addresses 0 to 9. That sometimes causes confusion and conflicts.

TzionCastillo1, 2022-02-16:
I have it set to 0x01, from what I've read this has seemed like the default address.

TzionCastillo1, 2022-02-17:
Update: It seems my issue was my power supply was causing the Y4000 to brown out.
Thank your for the help!

closed 2022-02-17 by TzionCastillo1