EnviroDIY/YosemitechModbus #25: Problems with the GetValues-TimingTest.ino

SteveClifto, 2022-03-09:
The code is failing right away with the model number not returning. I have tried this on a Y510 and a Y504, and both times I got the same output to the serial monitor (put below). I am using the example code and have only changed the model number and the DEREPin to the respective numbers for the code. I am using an Arduino Uno with pins 9 being Tx, 8 being Rx, and 5 being DEREPin. I wired the voltage and grounds separate instead of putting 5V to a pin, because the Y504 needs more than 5V to operate correctly. Here is a [link](https://www.amazon.com/dp/B00NIOLNAG?psc=1&ref=ppx_yo2_dt_b_product_details) to the TTL to rs485 converter I am using incase you're wondering.

Yosemitech  sensor for 
Waiting for sensor and adapter to be ready.
Getting sensor version.
    Current Hardware Version: 0.00
    Current Software Version: 0.00
Getting sensor serial number.
    Serial Number: 
Getting sensor calibration equation.
    Current Calibration Equation: final = 0.00*raw + 0.00
Starting sensor measurements
    Failed to start measuring!
Time(ms)  Temp(°C)  ()    Value
22388      -9999.00      -9999.00      -9999.00
28306      -9999.00      -9999.00      -9999.00

SRGDamia1, 2022-03-16:
Honestly, I've never successfully used the MAX485.  All of the boards I use are 3.3V and by the time you add in level shifting and manually setting flow directions, the timing is too off for stable communication.

Even if you're using different power supplies, make sure all of the ground planes are connected to each other. Are you using SoftwareSerial?  If so, try switching to AltSoftSerial; it's a better Serial library.  If you can, I'd get a RS485 chip with automatic flow control.

aufdenkampe, 2023-12-21:
Closing this issue, given that the RS485 converter or power supply were the likely culprits. Let us know if you found otherwise and still need help.

closed 2023-12-21 by aufdenkampe