EnviroDIY/YosemitechModbus #32: V4.0 specified in ModularSensors but not released

neilh10, 2022-12-27:
For enviroDIY/ModularSensors (develop)  library.json specifies
      "name": "YosemitechModbus",
      "owner": "envirodiy",
      "library id": "2078",
      "url": "https://github.com/EnviroDIY/YosemitechModbus",
      "version": "~0.4.0",

When trying to build I get 
Library Manager: Installing envirodiy/YosemitechModbus @ ~0.4.0
Library Manager: **Warning!** Could not install `{'name': 'YosemitechModbus', 'owner': 'envirodiy', 'library id': '2078', 'url': 'https://github.com/EnviroDIY/YosemitechModbus', 'version': '~0.4.0', 'note': 'Arduino library for communication with Yosemitech sensors via Modbus.', 'authors': ['Sara Damiano', 'Anthony Aufdenkampe'], 'frameworks': ['arduino'], 'platforms': ['atmelavr', 'atmelsam']}` dependency for the`EnviroDIY_ModularSensors` package

as   EnviroDIY/YosemitechModbus (master) doesn't have a 0.4.0  
and needs referencing to (develop) for it to work

Library Manager: Installing git+https://github.com/EnviroDIY/YosemitechModbus#develop

neilh10, 2023-01-25:
Others having a similar issue https://www.envirodiy.org/topic/yosemite-library-cant-be-find/#post-17555

aufdenkampe, 2023-01-25:
@neilh10, thanks for posting this issue!

I had been waiting merge in v0.4 here:
- #31 

Until after @SRGDamia1 to issued a new release of Modular sensors with this Pull Request:
- https://github.com/EnviroDIY/ModularSensors/pull/424

In hindsight, my work on YosemitechModbus v0.4 was designed to be backwards compatible, so I'll move forward with PR #31.

aufdenkampe, 2023-01-25:
@neilh10, I just merged PR#31 and issued a release.

PlatformIO's Library Registry should auto-update in the next day or so. See:
https://registry.platformio.org/libraries/envirodiy/YosemitechModbus

Once it shows v0.4.0, your problem should be fixed. Let me know.

neilh10, 2023-01-26:
thanks for the update - will try it. 

From past projects being clear how code is promoted takes a bit of thinking about, and there needs to be sand box to play around in. Hard to get it perfect straight away.

I think of (develop)  https://github.com/EnviroDIY/ModularSensors/tree/develop   as being where everything is worked on, prototyped, so slightly risky.   

So that was the case in just finding that 
as EnviroDIY/YosemitechModbus (master) doesn't have a 0.4.0
and needs referencing to EnviroDIY/YosemitechModbus (develop) for it to work

I typically try and merge from EnviroDIY/ModularSensors (master)   to my development branch 
neilh10/ModularSensors (rel1_dvlp1m) and then test there.
However in this case, because there was a lot on EnviroDIY/ModularSensors (develop),    so I pulled in the updates onto my development branch neilh10/ModularSensors (rel1_dvlp1m),  and the lib reference broke it, 
but was easy to work around.

Many thanks

neilh10, 2023-02-08:
Testing in my environment - works to pull from     https://github.com/EnviroDIY/YosemitechModbus

closed 2023-02-08 by neilh10