EnviroDIY/YosemitechModbus #35: Problem - No communication with Y14 Chlorophyll sensor with brush

primus192, 2023-06-15:
Hello. I am new to this sensors. We ordered DO, Conductivity, pH, Depth, and Chlorophyll sensor.

All of this sensors are working great at variety of voltages 5-12V except chlorophyll. I can't get any output from this sensor, no version, no serial number, no measurements, nothing.
Chlorophyll (not working):
[image]
Conductivity (working):
[image]

I am using our dedicated ESP32 board with rs485 adapter with DE, RE pins. I tried to adjust voltage from 5-12V, no luck. All grounds are common and connected together.

My sensor has HW 1.0 and SW 1.7 so i assume it is quite new. I reached out to Zoe from yosemitech. She sent me smart PC(2.6) software (a.k.a. Modbus runner?) and she told me to use USB-RS485 adapter so i bought one.

Strangely this chlorophyll sensor is working great with this USB-RS485 adapter even when I'm powering this sensor directly from it. I can read measurements, set, get, brush timings and manually activate brush.

[image]

[image]

There comes my question. What can i do to make this sensor work with my board? Change RS485 adapter to auto-direction controlled one? Add capacitors? Maybe someone had same problem here? I am attaching some pictures for reference.

SRGDamia1, 2023-06-15:
I've had much better luck in general with auto-direction RS485 adapters, but I don't have any experience with the chlorophyll sensor.

primus192, 2023-06-15:
> I've had much better luck in general with auto-direction RS485 adapters, but I don't have any experience with the chlorophyll sensor.

Thank you for Your attention. I have already ordered auto direction adapters to test that. I will get back with results when i test it although it is strange that other sensors are working fine on standard adapters but not chlorophyll.

primus192, 2023-06-16:
Ok my auto-direction RS485 adapter. It is working now. Closing as completed.

closed 2023-06-16 by primus192